micro-ROS/micro_ros_raspberrypi_pico_sdk #26: No connection/execution with the raspberry

kappa95, 2021-04-21:
I am trying to executing the example: `pico_micro_ros_example.c` following this [guide](https://ubuntu.com/blog/getting-started-with-micro-ros-on-raspberry-pi-pico)
The building seems ok: it returns from visual studio code this: 
```
bash
[proc] Executing command: /usr/local/bin/cmake --build /home/kappa95/micro_ros_ws/src/micro_ros_raspberrypi_pico_sdk/build --config Debug --target all -j 10 --
[build] [  1%] Performing build step for 'ELF2UF2Build'
[build] [  3%] Built target bs2_default
[build] [  6%] Built target bs2_default_padded_checksummed_asm
[build] [100%] Built target elf2uf2
[build] [  7%] No install step for 'ELF2UF2Build'
[build] [  8%] Completed 'ELF2UF2Build'
[build] [ 15%] Built target ELF2UF2Build
[build] Consolidate compiler generated dependencies of target pico_micro_ros_example
[build] [100%] Built target pico_micro_ros_example
[build] Build finished with exit code 0

```

I generate the uf2 file and I have uploaded it as in the guide. 
I have followed all the next steps of the guide and after: 
`micro-ros-agent serial --dev /dev/ttyACM0 baudrate=115200`

I get: 
```bash
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[1619029391.389393] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

```

The raspberry pico doesn't blink and I don't find the topics and nodes.

I am using ROS2 Dashing with Ubuntu 18.04.5 LTS.

Acuadros95, 2021-04-22:
Hello @kappa95, I have replicated the issue and it is related to connection timing.

The function `rclc_support_init` tries to connect with the agent with a timeout of 10 seconds, if the agent is not reachable in that time the scripts fails, not even reaching the LED activation. You can check this with the return value of the function, which returns false `(RCL_RET_OK)` on a successful connection.

I have modified the example to solve this problem (PR #29), now it will wait for a reachable agent for up to 2 minutes using the function `rmw_uros_ping_agent`. This should give enough time to start the agent and establish the connection.

Please give it a try and let us know the results

kappa95, 2021-04-22:
> Hello @kappa95, I have replicated the issue and it is related to connection timing.
>
> The function `rclc_support_init` tries to connect with the agent with a timeout of 10 seconds, if the agent is [...]

I pull the repo, I have seen the changes in the code. But I get the same issue

Acuadros95, 2021-04-23:
Could you please try with the dockerized foxy agent?

`docker run -it -v /dev:/dev --privileged  microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

kappa95, 2021-04-23:
> Could you please try with the dockerized foxy agent?
> 
> `docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

Same results:
```bash
docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1619164483.342955] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1619164483.343359] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

```

Second terminal:

```bash
ros2 node list
//
```

Acuadros95, 2021-04-23:
Could you check the serial port output without the agent, using miniterm or similar?

Disconnect and reconnect the board, then for the first 2 minutes there should appear a periodic message each second, like this:

```
sudo miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	
��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀	��␂␀␀␀�␒~␀␀␐␀�␀␀␀␂␀␀
```

If nothing appears, check that your raspberry pico is actually on the /dev/ttyACM0 port, by disconnecting it and checking that it is not on the /dev/tty list using `ls /dev/ttyACM*`.

kappa95, 2021-04-23:
> Could you check the serial port output without the agent, using miniterm or similar?
>
> Disconnect and reconnect the board, then for the first 2 minutes there should appear a periodic message each [...]

With miniterm, nothing appear: no messages.
And the only device is the raspberry: 
```bash 
ls /dev/ttyACM*
/dev/ttyACM0

```

[image]

I tried the hello world program from the raspberry site and using minicom worked.

Acuadros95, 2021-04-23:
After some investigation, the problem relies on the arm-none-eabi-gcc installed version: Ubuntu 18.04 uses 6.3.1, while the micro-ros library is compiled using 9.2.1. To check your actual version: `arm-none-eabi-gcc --version`

This commands will download a compatible version from source and set the cmake compiler path.

```
wget --directory-prefix=$HOME https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2
tar -xvf $HOME/gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2 -C $HOME
echo "export PICO_TOOLCHAIN_PATH=$HOME/gcc-arm-none-eabi-9-2020-q2-update" >> ~/.bashrc
source ~/.bashrc
```

Anyway, check the version output on the cmake command is at least 9.x.x:

```
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
```

kappa95, 2021-04-23:
> After some investigation, the problem relies on the arm-none-eabi-gcc installed version: Ubuntu 18.04 uses 6.3.1, while the micro-ros library is compiled using 9.2.1. To check your actual version: [...]

Thank you so much! Something now is changed:
```bash
docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1619178075.701850] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1619178075.702175] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1619178075.714481] info     | Root.cpp           | create_client            | create                 | client_key: 0x2B5411CA, session_id: 0x81
[1619178075.714548] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x726929866, address: 0
[1619178075.714607] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2B5411CA, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1619178075.717420] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2B5411CA, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2B 54 11 CA 81 00 FC 01
[1619178075.717486] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x726929866, address: 0
[1619178075.717542] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2B5411CA, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1619178075.719486] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2B5411CA, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2B 54 11 CA 81 00 FC 01
[1619178075.719580] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x726929866, address: 0
[1619178075.719652] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2B5411CA, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1619178075.824008] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2B5411CA, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2B 54 11 CA 81 00 FC 01
[1619178075.824408] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x726929866, address: 0
[1619178075.824503] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2B5411CA, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

```
The Led on the Raspberry now is on
But still: 
```bash
ros2 node list
//

```

Acuadros95, 2021-04-26:
Thank you for the feedback!

closed 2021-04-26 by Acuadros95

lsirikh, 2022-05-09:
I uses Ubuntu 20.04 and ros2 foxy. Is it possible to run the example code correctly?