micro-ROS/micro_ros_raspberrypi_pico_sdk #54: Low USB performance

errolt, 2021-05-06:
The USB performance seems to be below expectation. 

If I call this before rmw_uros_ping_agent:
```
    stdio_init_all();
    sleep_us(1000000);
    pico_serial_transport_write(0,"Hello, world2!\n",14,&b[0]);
```
then a 14 byte packet is sent in 250us. Looking at the USB packets the packet is split into multiple 2 byte USB packets. Not ideal, but not too bad.

Once rmw_uros_ping_agent is called, the PING messages take 8ms to 10ms to send. This is horrible, and the default int32 topic takes 13ms total to send, TX+RX.

It appears like there is an interrupt blocking the USB driver task, which is running on interrupt 100, but this is only speculation.

As a result, calling this before rmw_uros_ping_agent:
```
    stdio_init_all();
    sleep_us(1000000);
    fwrite("Hello, world3!\n",1,15,stdout);
```
results in the whole packet being split over two USB packets, and the fwrite returning in 10us. Much better.

But if I try and change pico_serial_transport_write to this:
```
size_t pico_serial_transport_write(struct uxrCustomTransport * transport, uint8_t *buf, size_t len, uint8_t *errcode)
{
    fwrite(buf,1,len,stdout);
    return len;
}
```
Then, due to the interrupt lockup, NO pings are sent every second, instead, after the 2 minute timeout, all the backed up data flushes out and you get a BUNCH of pings all in one go.

pablogs9, 2021-05-07:
Hello, @errolt thanks a lot for this research. Honestly, I'm not super in touch with the Raspberry Pi Pico SDK API so I guess that I should investigate more about the implementation of this transport when I have a time slot.

It would be great that you contribute an optimized USB transport for this platform.

CC: @lukicdarkoo do you have any tips on that?

errolt, 2021-05-07:
Hello @pablogs9. This is my first pico project, and my first uRos project, so I'm in deep. Don't worry about this issue. I was just hoping something obvious pops into your mind.

From my reading, both the Pico SDK built in USB stack and the optional TinyUSB stacks has a task that must be called on a regular bases. 
It looks like the built in USB stack sets up a timer that triggers a software interrupt, and the handler on that interrupt services the USB stack.
The TinyUSB stack has a function that must be called whenever spinning.

I've looked through the uros library and can't find an obvious issue, so I will need to connect a debugger to try and find the issue... Will post back when I have more info.

pablogs9, 2021-05-07:
Ok, thanks

pablogs9, 2022-01-28:
Any update on this?

errolt, 2022-01-28:
Unfortunately not. 

If I recall correctly then I was unable to debug the issue as micro_ros was just a binary blob, and I was unable to see what got broken when rmw_uros_ping_agent was called.

Due to this issue I had to abandon micro_ros as I was hoping to send motor encoder values over USB at high speed.

closed 2022-01-28 by errolt