micro-ROS/micro_ros_raspberrypi_pico_sdk #71: Subscriber running on pico not getting messages from a publisher node

JCDevSpace, 2021-05-20:
I am trying to setup a basic ping pong example to experiment with running a ros2 publisher and subscriber on a raspberry pico. Following the example, I setup a node running on the pico to make a int32 subscriber listening on the ping topic, then in the callback function incrementing the received message by one and publishes the results to a pong topic and have the on board LED flash between receiving the message and publishing it back.  Then on my laptop(ubuntu 20.04) I setup a node that listens on a pong topic and subscribes to the ping topic doing the same thing as the node running on the pico but in reverse. This in theory should allow the ping pong to go on forever once I publish a message to either one of the two topics from the command line. But I notice that the call back of the subscriber running on pico doesn't get trigger by messages published by other nodes, but responses to message published from the command line. And I made a different node doing the same thing as what was running on the pico on my laptop, and the ping ponging happens as expected when I run those two nodes. I have been searching for a while and have not found the reason for this. The following are the code setup I have right now:
running on the pico:
`#include <stdio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
#include <rmw_microros/rmw_microros.h>

#include "pico/stdlib.h"
#include "pico_uart_transports.h"

const uint LED_PIN = PICO_DEFAULT_LED_PIN;

rcl_publisher_t publisher;
rcl_subscription_t subscriber;
std_msgs__msg__Int32 msg;

void ping_cb(const void * msgin) {
  std_msgs__msg__Int32 * ping_num = (std_msgs__msg__Int32 *)msgin;
  ping_num->data++;
  gpio_put(LED_PIN, 0);
  sleep_ms(2000);
  rcl_publish(&publisher, ping_num, NULL);
  gpio_put(LED_PIN, 1);
}

int main() {
  rmw_uros_set_custom_transport(
		true,
		NULL,
		pico_serial_transport_open,
		pico_serial_transport_close,
		pico_serial_transport_write,
		pico_serial_transport_read
	);

  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  rcl_node_t node;
  rcl_allocator_t allocator;
  rclc_support_t support;
  rclc_executor_t executor;

  std_msgs__msg__Int32 msg;

  allocator = rcl_get_default_allocator();

  rcl_ret_t ret = rmw_uros_ping_agent(1000, 120);
  if (ret != RCL_RET_OK) {
    gpio_put(LED_PIN, 0);
    return ret;
  }

  rclc_support_init(&support, 0, NULL, &allocator);
  rclc_node_init_default(&node, "ping_pong", "pico", &support);
  
  rclc_publisher_init_default(
    &publisher, 
    &node, 
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "pong"
  );
  
  rclc_subscription_init_default(
    &subscriber,
    &node,
[... 221 lines cut ...]
  rclc
  std_msgs
  rmw_microxrcedds
)

install(TARGETS
  host_ping_pong
  host_pong_ping
  host_pong_ping_c
  DESTINATION lib/${PROJECT_NAME})

ament_package()
`

Running the micro ros agent when the pico is plugged in:
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1621472835.716347] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1621472835.716750] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1621472837.398044] info     | Root.cpp           | create_client            | create                 | client_key: 0x32A90DEE, session_id: 0x81
[1621472837.398085] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x849939950, address: 0`

LED comes on after starting the agent.

rqt_graph:
[image]

pablogs9, 2021-05-20:
A couple of things:
1. Can you share the output of the micro-ROS agent but using the verbose mode: `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`?
2. It is not recommended to sleep 2 seconds inside the subscription callback, it will freeze the executor and the middleware operation. Can you try without that? If you need to delay the publication, probably the best option is to setup a timer and handle the time difference between the subscription and the publication.

pablogs9, 2021-05-20:
One more thing! I'm seeing that your `rqt_graph` is seeing your micro-ROS node as `RTPSPartitipant`. This is a bug that has been solved (https://github.com/eProsima/Micro-XRCE-DDS/pull/92, https://github.com/micro-ROS/rmw-microxrcedds/pull/134, https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/250 and https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/246).

I have just updated the Raspberry Pico + micro-ROS libraries to include [these changes](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/pull/72), can you update your library?
**Also is required to rebuild the micro-ROS Agent.**, you also can use the dockerized version:

```
docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
```

If everything works well your `rqt_graph` should see the micro-ROS node as `/pico/ping_pong`

JCDevSpace, 2021-05-20:
Hi @pablogs9  thanks a lot for the feedback, I got rid of the sleeps as you suggested and ran the agent with the -v6 flag, the following is the outputs I got. So far, the behavior haven't changed.

[micro_ros_agent_v6_output.log](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/files/6519169/micro_ros_agent_v6_output.log)

Here's the modified code running on the pico:
```
#include <stdio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
#include <rmw_microros/rmw_microros.h>

#include "pico/stdlib.h"
#include "pico_uart_transports.h"

const uint LED_PIN = PICO_DEFAULT_LED_PIN;

rcl_publisher_t publisher;
rcl_subscription_t subscriber;
std_msgs__msg__Int32 msg;

void ping_cb(const void * msgin) {
  std_msgs__msg__Int32 * ping_num = (std_msgs__msg__Int32 *)msgin;
  ping_num->data++;
  gpio_put(LED_PIN, 0);
  rcl_publish(&publisher, ping_num, NULL);
  gpio_put(LED_PIN, 1);
}

int main() {
  rmw_uros_set_custom_transport(
		true,
		NULL,
		pico_serial_transport_open,
		pico_serial_transport_close,
		pico_serial_transport_write,
		pico_serial_transport_read
	);

  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);

  rcl_node_t node;
  rcl_allocator_t allocator;
  rclc_support_t support;
  rclc_executor_t executor;

  allocator = rcl_get_default_allocator();

  rcl_ret_t ret = rmw_uros_ping_agent(1000, 120);
  if (ret != RCL_RET_OK) {
    gpio_put(LED_PIN, 0);
    return ret;
  }

  rclc_support_init(&support, 0, NULL, &allocator);
  rclc_node_init_default(&node, "ping_pong", "pico", &support);
  
  rclc_publisher_init_default(
    &publisher, 
    &node, 
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "pong"
  );
  
  rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "ping"
  );

  executor = rclc_executor_get_zero_initialized_executor();
  rclc_executor_init(&executor, &support.context, 2, &allocator);
  rclc_executor_add_subscription(&executor, &subscriber, &msg, &ping_cb, ON_NEW_DATA);

  gpio_put(LED_PIN, 1);

  rclc_executor_spin(&executor);

  rcl_subscription_fini(&subscriber, &node);
  rcl_publisher_fini(&publisher, &node);
  rcl_node_fini(&node);

  gpio_put(LED_PIN, 0);

  return 0;
}
```
I will also pull the docker image and run with that to see what I get.

JCDevSpace, 2021-05-20:
No topic or nodes show up when the agent is ran with docker.

pablogs9, 2021-05-21:
Have you updated the library to the latest version in this repo?

JCDevSpace, 2021-05-21:
Thanks a lot, pulling the latest update on the main branch seem to have fixed the issue. Now I see the node running on pico as ping_pong and the ping ponging happening as expected.

closed 2021-05-21 by JCDevSpace