micro-ROS/micro_ros_raspberrypi_pico_sdk #83: Adding Custom Message

melearp, 2021-05-24:
Ordinarily I can either solve these problems myself, or find someone who has already done so. But this one has me stumped. This is the only mechanism I can find to pose the problem.

I have everything else installed and working and my own custom application working just fine between a Pico and a Ubuntu 20.04 machine. But I now need to add a custom message. I can do that just fine on the host side, but everything I try on the Pico microros side just fails. I even tried adding a simple custom message to the example application and that fails in the same way. No amount of fiddling with paths and CMake has got me anywhere.

The Microros tutorial for adding a custom message does not seem to apply correctly to the Pico installation. The very first step of the custom message tutorial says add
`find_package(rosidl_default_generators REQUIRED)`
to CMakeLists.txt

Even with this just one addition, running cmake .. in the build folder results in rosidl_default_generators not being found:
`CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findrosidl_default_generators.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rosidl_default_generators", but CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_default_generators" with any of the following names:

    rosidl_default_generatorsConfig.cmake
    rosidl_default_generators-config.cmake

  Add the installation prefix of "rosidl_default_generators" to
  CMAKE_PREFIX_PATH or set "rosidl_default_generators_DIR" to a directory
  containing one of the above files.  If "rosidl_default_generators" provides
  a separate development package or SDK, be sure it has been installed.`

I tried putting in the paths, as CMake demands, but this just digs a deeper hole.
Are there some instructions somewhere that work for this Pico fork??

pablogs9, 2021-05-25:
What I have done is:

1. Adding this folder[my_custom_message.zip](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/files/6536895/my_custom_message.zip) to `
microros_static_library/library_generation/extra_packages`. This folder is created using [this instructions](https://micro.ros.org/docs/tutorials/advanced/create_new_type/).
2. Rebuild the micro-ROS static library for Raspberry Pico using 
`docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:foxy`

It build correctly, let me know if you are able to solve this with this steps.

melearp, 2021-05-25:
For the docker command

docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:foxy

1. I presume this is run in the directory which contains the microros directory
2. Is it necessary to change "project" to something that I have locally or is it actually the word "project"

I don't use docker much so am not familiar with its usage and actions.

Thanks for the very quick response.

pablogs9, 2021-05-25:
You just need to run this command, **without modification** in your project folder. In the case you are building the project provided with this repo, just in the repo folder:

```bash
git clone https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk
cd micro_ros_raspberrypi_pico_sdk
docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:foxy
```

melearp, 2021-05-25:
I have a way forward now.
Thank you.

closed 2021-05-25 by pablogs9