micro-ROS/micro_ros_raspberrypi_pico_sdk #98: pico_node can't be found and Serial Exception: device reports readiness to read but returned no data

cacticaxas, 2021-06-02:
I'm trying to run the example from the README but the pico_node can't be found.

The problems are similar to the post (https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/26) from kappa 95 and my cmake command returns

```
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
```

however, when the code is run the minterm of the of the port would receive this SerialException error, where it says the device is ready to read but there is no data returned.
```

caxas@caxas-Lenovo-ideapad-700-15ISK:~/micro_ros_raspberrypi_pico_sdk/build$ sudo miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␀␀␀␀�Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 500, in read

--- exit ---
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/bin/miniterm", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read
```
And of course ros2 node list would return nothing, although the pico is running.

```
caxas@caxas-Lenovo-ideapad-700-15ISK:~/micro_ros_raspberrypi_pico_sdk/build$ sudo micro-ros-agent serial --dev /dev/ttyACM0 -b 115200
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[1622642059.282302] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1622642062.136104] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1622642062.136149] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 2
[1622642078.640980] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 1
[1622642079.141126] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 13. Please re-run with superuser privileges.
[1622642079.646925] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1622642080.148194] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1622642080.149290] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1622642080.149310] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1622642080.150302] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
```
Assistance on this would be much appreciated. Thanks in advance.

pablogs9, 2021-06-03:
Can you share the output of the micro-ROS Agent with the flag `-v6`? Do not open the micro-ROS Agent and minicom simultaneously.

cacticaxas, 2021-06-08:
Hi pablogs9, this is currently the output I'm getting,

`caxas@caxas-Lenovo-ideapad-700-15ISK:~$  docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1623159026.064243] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1623159026.064512] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
`
In this case only the micro-ROS Agent is being run and nothing else.

The ros2 node list returns nothing so far.

Apologies as well for the late reply, I was away for awhile.

cacticaxas, 2021-06-08:
*UPDATE*

I run the code again and I got this

`caxas@caxas-Lenovo-ideapad-700-15ISK:~$ docker run -it -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6
[1623159545.762984] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1623159545.763237] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1623159557.662916] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 4
[1623159557.673038] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159558.677549] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159559.682923] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159560.688567] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159560.911450] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1623159585.068894] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 4
[1623159585.069107] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 19
[1623159585.079219] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159586.081590] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159587.085418] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159588.090018] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1623159589.048011] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1623159596.696098] info     | Root.cpp           | create_client            | create                 | client_key: 0x49C6B4EB, session_id: 0x81
[1623159596.696221] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1237759211, address: 0
[1623159596.696411] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1623159597.626446] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159597.626916] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159597.627626] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159597.627897] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159597.805078] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159597.805383] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159598.908706] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159598.908990] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159599.053981] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159599.054265] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159599.233099] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159599.233296] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1623159599.243502] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1623159599.243725] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[... cut ...]
[1623159601.201318] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x49C6B4EB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80`

The peculiar things is I did not change anything in the .uf2 file and I'm running the same command as before.

and I at least can find the node now

`caxas@caxas-Lenovo-ideapad-700-15ISK:~$ ros2 node list
/pico_node
`

but the topic called /pico_publisher is not appearing

 `caxas@caxas-Lenovo-ideapad-700-15ISK:~$ ros2 topic list
/image
/parameter_events
/rosout
`

pablogs9, 2021-06-09:
Try to use the agent this way:

`docker run -it -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

cacticaxas, 2021-06-09:
Thank you so much!!! Running the docker with 

`docker run -it -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

fixed it.

It's running again.

closed 2021-06-09 by cacticaxas