micro-ROS/micro_ros_raspberrypi_pico_sdk #130: Unreliable operations including core dump

UAV-Pilot, 2021-06-27:
My host machine is a Jetson Nano running Ubuntu 18.04.5 LTS. ROS2 galactic and micro_ros_agent were built from source using gcc 9.4. I encountered 2 issues: 1) most of time pico_node did not show up 2) about 2/3 of times micro_ros_agent core dumped. 

pico_node issue:
-----------------------
When pico_node did show up, it only showed up in the 1st call after stating micro_ros_agent as shown below:

rfu@NanoProto:~$ ros2 node list
/pico_node
rfu@NanoProto:~$ ros2 topic list
/parameter_events
/pico_publisher
/rosout
rfu@NanoProto:~$ ros2 topic echo /pico_publisher
1624826059.101388 [0]       ros2: using network interface wlan0 (udp/192.168.1.205) selected arbitrarily from: wlan0, mpqemubr0, lxdbr0
data: 36
---
data: 37
---
data: 38
---
^Crfu@NanoProto:~$ ros2 node list
Traceback (most recent call last):
  File "/home/rfu/src/ros2_galactic/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/home/rfu/src/ros2_galactic/build/ros2cli/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/rfu/src/ros2_galactic/build/ros2node/ros2node/command/node.py", line 37, in main
    return extension.main(args=args)
  File "/home/rfu/src/ros2_galactic/build/ros2node/ros2node/verb/list.py", line 38, in main
    node_names = get_node_names(node=node, include_hidden_nodes=args.all)
  File "/home/rfu/src/ros2_galactic/build/ros2node/ros2node/api/__init__.py", line 60, in get_node_names
    node_names_and_namespaces = node.get_node_names_and_namespaces()
  File "/usr/lib/python3.6/xmlrpc/client.py", line 1112, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.6/xmlrpc/client.py", line 1452, in __request
    verbose=self.__verbose
  File "/usr/lib/python3.6/xmlrpc/client.py", line 1154, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.6/xmlrpc/client.py", line 1170, in single_request
    return self.parse_response(resp)
  File "/usr/lib/python3.6/xmlrpc/client.py", line 1342, in parse_response
    return u.close()
  File "/usr/lib/python3.6/xmlrpc/client.py", line 656, in close
    raise Fault(**self._stack[0])
xmlrpc.client.Fault: <Fault 1: "<class 'rclpy._rclpy_pybind11.RCLError'>:Failed to get node names: empty node name returned by the RMW layer, at /home/rfu/src/ros2_galactic/src/ros2/rcl/rcl/src/rcl/graph.c:360">

BTW, when "ros2 node list" failed, "ros2 topic echo /pico_publisher" still output received data continuously with the received number increasing sequentially.

core dump issue
-----------------------
Core dump happened after 4 or 5 recv_message printouts as shown below:

rfu@NanoProto:~/src/micro_ros_ws/install/micro_ros_agent/lib/micro_ros_agent$ ./micro_ros_agent serial --dev /dev/ttyACM0 baudrate=115200 -v6
[1624825772.540376] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1624825772.540878] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1624825777.023673] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1624825777.033836] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825778.035949] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825779.036863] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825780.038924] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825781.040222] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825782.042234] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1624825782.351269] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1624825784.367612] info     | Root.cpp           | create_client            | create                 | client_key: 0x5A8E7986, session_id: 0x81
[1624825784.367729] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1519286662, address: 0
[1624825784.367912] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5A8E7986, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1624825785.267536] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5A8E7986, len: 40, data: 
0000: 81 80 00 00 01 05 20 00 00 0A 00 01 01 03 00 00 12 00 00 00 00 01 2D 00 0A 00 00 00 70 69 63 6F
0020: 5F 6E 6F 64 65 00 00 00
[... 32 lines cut ...]
[1624825915.403924] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0E491A7D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1624825915.411565] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0E491A7D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
Segmentation fault (core dumped)

----------------------------------------------------------------

When the core dump did not happen at the beginning, micro_ros_agent could run for a log time, so it looks like once data/objects  were initialized correctly, core dump would not happen. For example, in the following run, the sequence in the message reached 6000:

^Crfu@NanoProto:~$ ros2 topic echo /pico_publisher
1624832020.121484 [0]       ros2: using network interface wlan0 (udp/192.168.1.205) selected arbitrarily from: wlan0, mpqemubr0, lxdbr0
data: 5997
---
data: 5998
---
data: 5999
---
data: 6000
---
data: 6001
---
data: 6002

BTW, as my intended application will not be run in docker, so any solution involving docker is not applicable in this case. Also upgrading to Ubuntu 20.x is not an option as it will not happen on Jetson devices for quite long time. I wish these issues could be resolved soon as currently I have no confidence in using it in the application.

pablogs9, 2021-06-28:
Can you share your client code so we can replicate the issue?
Which version of ROS 2 are you using? Which middleware are you using? Which version of micro-ROS client and agent are you using?

UAV-Pilot, 2021-06-28:
I'm in the evaluation phase, so no custom code was involved yet. The problem was encountered in the course of trying to replicate what's described here: https://ubuntu.com/blog/getting-started-with-micro-ros-on-raspberry-pi-pico. ROS2 galactic was cloned and built within the last 3 days (not sure the exact version) according to steps described here: https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html. 

Middleware source code and micro-ROS source code were cloned within the last 3 days according to steps described here: https://github.com/micro-ROS/micro_ros_setup. Although the provided setup scripts were not usable on Jetson Nano, but I should have ended up with the same set of source code. Some of directories are:

~/src/micro_ros_ws/src/eProsima/Micro-XRCE-DDS-Agent (and ...-Client)

~/micro_ros_ws/src/micro_ros_raspberrypi_pico_sdk

~/src/micro_ros_ws/src/uros$ 
     drive_base/
     micro-ROS-Agent/
     micro-ROS-demos/
     micro_ros_msgs/
     micro_ros_utilities/
     rclc/
     rmw_microxrcedds/
     rosidl_typesupport_microxrcedds/

So the agent is micro_ros_ws/src/uros/micro-ROS-Agent, and client was from https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk.git.

pablogs9, 2021-06-28:
First of all, it seems that you are using ROS 2 Galactic, the default middleware of ROS 2 Galactic is not fully supported in micro-ROS. You can start by changing it to Fast-DDS: https://docs.ros.org/en/galactic/Guides/Working-with-multiple-RMW-implementations.html

UAV-Pilot, 2021-06-29:
@pablogs9  Changing the middleware to Fast-DDS solved the problem. Thanks a lot for your help!

BTW, when sourcing ROS2 and micro-ROS local setup scripts, which one do your source first?

pablogs9, 2021-06-29:
first ROS2 and then micro-ROS

closed 2021-06-29 by pablogs9