micro-ROS/micro_ros_raspberrypi_pico_sdk #149: Subscribing topics not works in micro-ros but publishing topics was OK (in raspberrypi-pico, teensy, STM32Cube) using static library of uros

masoudir, 2021-07-11:
## Issue template

- Hardware description: RaspberryPi-Pico, Also Teensy 4.0, Also STM32Cube for STM32 microcontrollers
- Version or commit hash: foxy

#### Steps to reproduce the issue
Just add the subscribing functions to the main code in this (file)[https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/galactic/pico_micro_ros_example.c]

#### Expected behavior

I have written a code that changes the value of integer that is publishing over 1 sec, using the value that must be subscribed from another topic. The value must be changed.

#### Actual behavior

But we see that subscrbing callback function does not happen. I see this problem in Raspberry-pico, Teensy, STM32Cube

#### Additional information

My main code is:
            
            #include <stdio.h>
            
            #include <rcl/rcl.h>
            #include <rcl/error_handling.h>
            #include <rclc/rclc.h>
            #include <rclc/executor.h>
            #include <std_msgs/msg/int32.h>
            #include <rmw_microros/rmw_microros.h>
            
            #include "pico/stdlib.h"
            #include "pico_uart_transports.h"
            
            const uint LED_PIN = 25;
            
            rcl_publisher_t publisher;
            
            rcl_subscription_t subscriber;
            
            std_msgs__msg__Int32 msg;
            
            void timer_callback(rcl_timer_t *timer, int64_t last_call_time)
            {
                rcl_ret_t ret = rcl_publish(&publisher, &msg, NULL);
                msg.data++;
            }
            
            void subscription_callback(const void * msgin)
            {  
              const std_msgs__msg__Int32 * msg_ = (const std_msgs__msg__Int32 *)msgin;
              if(msg_->data > 0){
                  msg.data = msg_->data;
              }
              
            }
            
            
            int main()
            {
                rmw_uros_set_custom_transport(
		            true,
		            NULL,
		            pico_serial_transport_open,
		            pico_serial_transport_close,
		            pico_serial_transport_write,
		            pico_serial_transport_read
	            );
            
                gpio_init(LED_PIN);
                gpio_set_dir(LED_PIN, GPIO_OUT);
            
                rcl_timer_t timer;
                rcl_node_t node;
                rcl_allocator_t allocator;
                rclc_support_t support;
                rclc_executor_t executor;
            
                allocator = rcl_get_default_allocator();
            
                // Wait for agent successful ping for 2 minutes.
                const int timeout_ms = 1000; 
                const uint8_t attempts = 120;
            
                rcl_ret_t ret = rmw_uros_ping_agent(timeout_ms, attempts);
            
                if (ret != RCL_RET_OK)
                {
                    // Unreachable agent, exiting program.
                    return ret;
                }
            
                rclc_support_init(&support, 0, NULL, &allocator);
            
                rclc_node_init_default(&node, "pico_node", "", &support);
                rclc_publisher_init_default(
                    &publisher,
                    &node,
                    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
                    "pico_publisher");
            
                rclc_subscription_init_default(
                &subscriber,
                &node,
                ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
                "micro_ros_subscriber");
                    
            
                rclc_timer_init_default(
                    &timer,
                    &support,
                    RCL_MS_TO_NS(1000),
                    timer_callback);
            
                rclc_executor_init(&executor, &support.context, 1, &allocator);
                rclc_executor_add_timer(&executor, &timer);
                rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA);
            
                gpio_put(LED_PIN, 1);
            
                msg.data = 0;
                while (true)
                {
                    rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
                }
                return 0;
            }

pablogs9, 2021-07-12:
Can you share the micro-ROS Agent output?

masoudir, 2021-07-14:
My uros agent messages:

      $ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
      [1626271348.710050] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
      [1626271348.710630] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
      [1626271350.418708] info     | Root.cpp           | create_client            | create                 | client_key: 0x67E96256, session_id: 0x81
      [1626271350.418969] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1743348310, address: 0
      [1626271351.582799] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x67E96256, participant_id: 0x000(1)
      [1626271351.601182] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x67E96256, topic_id: 0x000(2), participant_id: 0x000(1)
      [1626271351.717542] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x67E96256, publisher_id: 0x000(3), participant_id: 0x000(1)
      [1626271351.735192] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x67E96256, datawriter_id: 0x000(5), publisher_id: 0x000(3)
      [1626271351.743218] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x67E96256, topic_id: 0x001(2), participant_id: 0x000(1)
      [1626271351.889874] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x67E96256, subscriber_id: 0x000(4), participant_id: 0x000(1)
      [1626271352.014744] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x67E96256, datareader_id: 0x000(6), subscriber_id: 0x000(4)

masoudir, 2021-07-14:
Also if I substitute the position of "add_timer" function and "add_subscription" function with each other, I lose topic publishing. It hangs at add_subscrption function.

In fact, if I use this:

            rclc_executor_init(&executor, &support.context, 1, &allocator);
            rclc_executor_add_timer(&executor, &timer);
            rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA);

I have publishing and no subscribing, and if I use this:

            rclc_executor_init(&executor, &support.context, 1, &allocator);
            rclc_executor_add_timer(&executor, &timer);
            rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA);

Then I have no topic publishing and no topic subscribing.

pablogs9, 2021-07-14:
@JanStaschulat would you be ok if I implement the executor handles as a double linked list so we can allocate them in the `rclc_executor_add_*` API and avoid initing the executor with a certain number of handles? I have seen this error some other times...

masoudir, 2021-07-15:
> You are initing the executor with only 1 handle and adding two entities:
> 
> ```c
> ... the "1" here:
> rclc_executor_init(&executor, &support.context, 1, &allocator);
> ```

Thanks. It has worked. I have changed this line:

     rclc_executor_init(&executor, &support.context, 1, &allocator);

To this:

    rclc_executor_init(&executor, &support.context, 2, &allocator);

Thanks in advanced

pablogs9, 2021-07-15:
Cool, I'm closing this.

Do not hesitate to open new issues if you have any problems.

closed 2021-07-15 by pablogs9