micro-ROS/micro_ros_raspberrypi_pico_sdk #228: Timer timeout bellow 500ms doesn't work

negre, 2021-09-22:
As mentionned in issue https://github.com/micro-ROS/micro_ros_arduino/issues/56 timer bellow 500ms doesn't work on the arduino nano rp2040 connect

## Issue template

- Hardware description: Arduino Nano RP2040 Connect
- RTOS:
- Installation type: Arduino with micro_ros_arduino
- Version or commit hash: galactic

#### Steps to reproduce the issue

1. copy the micro-ros_publisher example and set the timer_timeout to 100 ms instead of 1000
```
$ diff orig.ino new.ino 
64c64
<   const unsigned int timer_timeout = 1000;
---
>   const unsigned int timer_timeout = 100;
79d78
<   delay(100);
```
2. start micro_ros agent
```
$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 115200
```
3. display topic frequency :
```
$ros2 topic hz /micro_ros_arduino_node_publisher
average rate: 1.577
	min: 0.101s max: 0.901s std dev: 0.37674s window: 3
average rate: 1.722
	min: 0.101s max: 0.901s std dev: 0.39155s window: 5
average rate: 1.792
	min: 0.101s max: 0.901s std dev: 0.39554s window: 7
average rate: 1.834
	min: 0.101s max: 0.901s std dev: 0.39716s window: 9
average rate: 1.861
	min: 0.101s max: 0.901s std dev: 0.39798s window: 11
```

#### Expected behavior

Average rate should be 10 Hz

#### Actual behavior

Average rate is 2 Hz

#### Additional information

Acuadros95, 2021-09-23:
Hello @negre,

The compiler version of the micro-ROS library builder for this board was outdated. This caused unexpected behavior on executor internal time measurements. This PR should fix the problem: https://github.com/micro-ROS/micro_ros_arduino/pull/490. 

I have updated the releases with this fix, please give it a try: https://github.com/micro-ROS/micro_ros_arduino/releases

Acuadros95, 2021-09-27:
Closing this, reopen if the problem persist

closed 2021-09-27 by Acuadros95