micro-ROS/micro_ros_raspberrypi_pico_sdk #308: Raspberry Pico - publisher node not coming up and publishing messages for example

robofoundry, 2021-11-29:
- Hardware description: Raspberry Pico 2021
- RTOS: Not sure. compiled the pico sdk per directions and uploaded the pico_micro_ros_example.uf2 to the Pico board
- Installation type: micro-ros - followed instructions to get and compile the code from article here which is same as readme file for this project - https://ubuntu.com/blog/getting-started-with-micro-ros-on-raspberry-pi-pico
- Version or commit hash: foxy

#### Steps to reproduce the issue
1. Get pico-sdk and compile it
2. get the example repo code and compile it [had to checkout from foxy branch as default galactic branch code did not compile]
3. Upload the file with command - cp build/pico_micro_ros_example.uf2 /media/$USER/RPI-RP2
4. Install snap for micro-ros and connect with serial port using following commands :
5. snap connect micro-ros-agent:serial-port snapd:pico
6. micro-ros-agent serial --dev /dev/ttyACM0 baudrate=115200

At this point the info message shows that agent is running:
[1638229049.680105] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

7. But when running "ros2 node list" or "ros2 topic list", it does not show any node is running
8. The article referenced above says that the initialization code in example is not perfect so try to plug/unplug multiple times and wait for 10 secs, tried it several times but did not light up the led or show any messages being published on ros2 topics.

#### Expected behavior
The ros2 node should show up and published messages should be captured by topic echo
#### Actual behavior
No node is running and no messages are being published.
#### Additional information
All code was compiled on Ubuntu 20.04 and ROS2 foxy

Acuadros95, 2021-11-30:
Hi @robofoundry,

The snap micro-ros-agent has problems using serial ports, could you try with the dockerized agent? Like this:
`docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6`

Acuadros95, 2021-12-21:
Closing this due to inactivity, reopen if the problem persist

closed 2021-12-21 by Acuadros95