micro-ROS/micro_ros_raspberrypi_pico_sdk #309: how to reconnect agent  without removal pico from computer

JuFengWu, 2021-11-30:
- Hardware description:  Pico
- Version or commit hash: Foxy

I tried to following the [example code(pico_micro_ros_example.c)](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/galactic/pico_micro_ros_example.c) , but I found when I close the ros-agent in my computer, and restart the ros-agent, it can not send command again.

How should I do?

Thank you.

Acuadros95, 2021-11-30:
Hello @JuFengWu,

Could you exactly describe your problem? That example is not prepared to reconnect to a closed agent.

If that is what you want, you can check [this other example](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) which reinitialize everything on agent restart.

JuFengWu, 2021-11-30:
Thank you.

I modified the your example code from Arduino to Pico, and it work to me. 

There is re-connect code for Pico and work for me

https://gist.github.com/JuFengWu/4eb0d8ce2aa5d0ce8758449d3cf2adfd

closed 2021-11-30 by JuFengWu

reinzor, 2023-07-13:
We also experienced issues with this; for us, the solution is now simply pinging the micro ros agent and reset the pico if this fails; sample code:

```
void software_reset()
{
  watchdog_enable(1, 1);
  while(1);
}

if (rmw_uros_ping_agent(10, 1) != RCL_RET_OK)
{
  // Unreachable agent, restart program.
  software_reset();
}
```