micro-ROS/micro_ros_raspberrypi_pico_sdk #345: Where is pico_micro_ros_example.uf2?

rsc9421, 2021-12-21:
I follow the instructions (3 times) but I cant find the file "pico_micro_ros_example.uf2" inside the build folder (of course after make)

pablogs9, 2021-12-22:
I have replicated the instructions and the file is in `build/pico_micro_ros_example.uf2`.

Could you provide more details?

Blebot0, 2022-01-21:
Even I am facing the same issue. I followed the steps as mentioned in README

pablogs9, 2022-01-21:
Could you provide more details? Steps, build output?

Blebot0, 2022-01-21:
I am using Fedora-35.
I installed all the packages using dnf and built **gcc-arm-none-eabi** from source.

Then I cloned the repository in my home directory. Made a build directory inside the pico-sdk dir. 

Then i ran 
`cmake ..`
and 
`make all `

Here is the output: 
~/pico-sdk/build> make all
[  0%] Built target bs2_default
[  0%] Performing build step for 'ELF2UF2Build'
Consolidate compiler generated dependencies of target elf2uf2
[100%] Built target elf2uf2
[  0%] No install step for 'ELF2UF2Build'
[  0%] Completed 'ELF2UF2Build'
[  1%] Built target ELF2UF2Build
[  1%] Built target bs2_default_padded_checksummed_asm
Consolidate compiler generated dependencies of target pico_stdlib_test
[  8%] Built target pico_stdlib_test
Consolidate compiler generated dependencies of target pico_time_test
[ 15%] Built target pico_time_test
Consolidate compiler generated dependencies of target pico_divider_test
[ 21%] Built target pico_divider_test
Consolidate compiler generated dependencies of target pico_divider_nesting_test_with_dirty_check
[ 28%] Built target pico_divider_nesting_test_with_dirty_check
Consolidate compiler generated dependencies of target pico_divider_nesting_test_with_disable_irq
[ 35%] Built target pico_divider_nesting_test_with_disable_irq
Consolidate compiler generated dependencies of target pico_float_test
[ 42%] Built target pico_float_test
Consolidate compiler generated dependencies of target pico_double_test
[ 49%] Built target pico_double_test
Consolidate compiler generated dependencies of target kitchen_sink
[ 57%] Built target kitchen_sink
Consolidate compiler generated dependencies of target kitchen_sink_extra_stdio
[ 69%] Built target kitchen_sink_extra_stdio
Consolidate compiler generated dependencies of target kitchen_sink_copy_to_ram
[ 78%] Built target kitchen_sink_copy_to_ram
Consolidate compiler generated dependencies of target kitchen_sink_no_flash
[ 86%] Built target kitchen_sink_no_flash
Consolidate compiler generated dependencies of target hardware_pwm_test
[ 93%] Built target hardware_pwm_test
Consolidate compiler generated dependencies of target cmsis_test
[100%] Built target cmsis_test

Blebot0, 2022-01-21:
I can rebuild it if it helps.

pablogs9, 2022-01-21:
have you tested `make` instead `make all`?

Blebot0, 2022-01-21:
Yeah

Blebot0, 2022-01-21:
CMAKE Output:

~/pico-sdk/build> cmake ..
PICO_SDK_PATH is /home/blebot/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/blebot/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.1") found components: Interpreter
TinyUSB available at /home/blebot/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/blebot/pico-sdk/build

pablogs9, 2022-01-21:
As you can see, in Ubuntu it is working in our CI: https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/runs/4891919559?check_suite_focus=true

Also, you have the steps here: https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/8dd69132bc16ec0d76f61e51a442ee26329b3b8f/.github/workflows/ci.yml#L36

Blebot0, 2022-01-21:
Ok! I'll compare the output and zero down on the issue

pablogs9, 2022-01-28:
Any update on this?

pablogs9, 2022-02-21:
Solved here: https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/pull/413

closed 2022-02-21 by pablogs9