micro-ROS/micro_ros_raspberrypi_pico_sdk #412: Issue Compiling Library

Ad353462, 2022-02-19:
- Hardware description: Ubuntu 20.04 / Pi Pico
- RTOS: N/A
- Version or commit hash: foxy

Hello, I'm not sure if this is an appropriate issue to raise, but I have attempted all the fixes I could find elsewhere online and haven't had any luck resolving my issue. I'm working on adding a custom message to the microros install. I tried following along with the steps provided in the "Adding Custom Message #83" issue raised prior, but ran into the issue below during compiling the library using docker.

#### Expected behavior
Library compiles successfully incorporating custom ros2 message type.

#### Actual behavior
While trying to compile libmicroros using: 
`docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:foxy`

In the git cloned micro_ros_raspberrypi_pico_sdk directory, I’m getting the error:

“/entrypoint.sh: line 9: /project/microros_static_library/library_generation/library_generation.sh: Permission denied”

#### Additional information
The docker image pulled using: `docker pull microros/micro_ros_static_library_builder:foxy` is up to date, and I’ve tried running the docker run command with --privileged and running with sudo, I tried reinstalling docker, and adding my user to the docker group, but I can’t seem to get it to work.

Any advice is appreciated. Thank you in advance.

pablogs9, 2022-02-21:
Maybe you can check the permissions for the file `library_generation.sh` and use chmod for adding the permission for all the users. In any case, default user for docker is root so this is strange, do you have any special configuration for docker?

```bash
➜  micro_ros_raspberrypi_pico_sdk git:(foxy) ls -la microros_static_library/library_generation
total 24
drwxr-xr-x 3 pgarrido pgarrido 4096 abr  7  2021 .
drwxr-xr-x 3 pgarrido pgarrido 4096 abr  7  2021 ..
-rwxr-xr-x 1 pgarrido pgarrido 1693 abr  7  2021 colcon.meta
drwxr-xr-x 2 pgarrido pgarrido 4096 abr  7  2021 extra_packages
-rwxr-xr-x 1 pgarrido pgarrido 2203 feb 21 07:34 library_generation.sh
-rwxr-xr-x 1 pgarrido pgarrido 1357 abr  7  2021 toolchain.cmake
➜  micro_ros_raspberrypi_pico_sdk git:(foxy) chmod 777 microros_static_library/library_generation/library_generation.sh 
➜  micro_ros_raspberrypi_pico_sdk git:(foxy) ls -la microros_static_library/library_generation                         
total 24
drwxr-xr-x 3 pgarrido pgarrido 4096 abr  7  2021 .
drwxr-xr-x 3 pgarrido pgarrido 4096 abr  7  2021 ..
-rwxr-xr-x 1 pgarrido pgarrido 1693 abr  7  2021 colcon.meta
drwxr-xr-x 2 pgarrido pgarrido 4096 abr  7  2021 extra_packages
-rwxrwxrwx 1 pgarrido pgarrido 2203 feb 21 07:34 library_generation.sh
-rwxr-xr-x 1 pgarrido pgarrido 1357 abr  7  2021 toolchain.cmake
➜  micro_ros_raspberrypi_pico_sdk git:(foxy) 
```

Ad353462, 2022-02-21:
I checked the permissions of the library_generation file with:

`ls -la <path to library_generation.sh>`

As recommended, and it was missing the -x permission. Adding it with:
`chmod +x <path to library_generation.sh>`

Seems to have resolved the issue. I was able to re-compile the library and the custom message type is now available in the environment.

 To answer your question though, no, I don’t have any special configurations configured for docker, to my knowledge anyway. I’ll have to take a closer look again, maybe I installed it improperly.

Regardless, thank you for the response. Apologies that it seems to have been a basic issue on my end, I'm not as familiar with docker as I would like to be.

Thanks again!

closed 2022-02-21 by Ad353462