micro-ROS/micro_ros_raspberrypi_pico_sdk #428: micro-ros-agents stops executing when connecting RPi Pico, gives "open device error"

sejego, 2022-03-04:
- Hardware description: Raspberry Pi Pico
- Installation type: cloned micro-ros from Github repo and built it. (e.g. i did not use snap/docker to run it)
- Version or commit hash: latest for **foxy**

#### Steps to reproduce the issue
1. Flash the example code on the Raspberry Pi Pico
2. Run the micro-ros-agent for serial output with
 `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 baudrate=115200`
3. connect Raspberry Pi Pico to PC  

#### Expected behavior
micro-ros-agent recognizes that there is a connection in serial and starts accepting messages
#### Actual behavior
Open device error and micro-ros-agent stops executing
#### Additional information
Provided log:
```
sejego@pop-os:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 baudrate=115200
[1646397373.328044] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 13. Please re-run with superuser privileges.
Error while starting serial agent!

```

Previously I was running the agent with connected Teensy 4.0, everything was working smoothly.

pablogs9, 2022-03-04:
`device: /dev/ttyACM0, errno: 13. Please re-run with superuser privileges.`

sejego, 2022-03-09:
Yeah, indeed changing rules on `/dev/tty/ACM0` fixes the problem. 
Is it a normal behaviour that Pico required such a step and other boards do not?

pablogs9, 2022-03-09:
I guess that it depends on your system permissions.

I'm closing, please feel free to reopen if you have further questions.

closed 2022-03-09 by pablogs9