micro-ROS/micro_ros_raspberrypi_pico_sdk #439: rmw_uros_epoch_millis() is returning a 32 bit integer istead of a 64 bit integer

leon3428, 2022-03-13:
## rmw_uros_epoch_millis() is returning a 32 bit integer istead of a 64 bit integer

- Hardware description: Raspberry Pi Pico
- RTOS: bare metal
- Installation type: cloned micro_ros_raspberrypi_pico_sdk repository
- Version or commit hash: latest galactic

#### Steps to reproduce the issue
Add the following code to the provided example
```c
// Sync timeout
const int timeout_ms = 1000;

// Synchronize time with the agent
rmw_uros_sync_session(timeout_ms);

if (rmw_uros_epoch_synchronized()) 
{
    // Get time in milliseconds
    int64_t time_ms = rmw_uros_epoch_millis();
}
```

#### Expected behavior
rmw_uros_epoch_millis() should return a 64 bit integer; time since the unix epoch in ms.

#### Actual behavior
Returned value is equal to the correct time after overflowing.
Example:
 - expected output: 1647191822823
 - actual output: -2075618841

#### Additional information
I need the exact time in order to sync tf transforms.

pablogs9, 2022-03-14:
`rmw_uros_epoch_millis()` is returning an `int64_t` as you can see [here](https://github.com/micro-ROS/rmw_microxrcedds/blob/51942f621a03a60f932f788a9c262e119dec3f99/rmw_microxrcedds_c/src/rmw_microros/time_sync.c#L37-L50).

How are you retrieving the value of `time_ms`?

leon3428, 2022-03-14:
After further looking into it, I noticed that I did not have ```#include <rmw_microros/rmw_microros.h>``` in the file that I was working on. ```rmw_uros_epoch_millis()``` was implicitly declared, and it assumed that the return type is int.

closed 2022-03-14 by leon3428