micro-ROS/micro_ros_raspberrypi_pico_sdk #467: Services not responding and giving garbage data in the requests

rjp5th, 2022-04-05:
## Issue template

- Hardware description: Raspberry Pi Pico
- RTOS: None
- Installation type: Prebuilt microros from micro_ros_raspberrypi_pico_sdk repo with host running Ubuntu 20.04.4 LTS
- Version or commit hash: galactic (5f13e4f)

#### Steps to reproduce the issue
1. Modify the pico_micro_ros_example.c with micro-ROS-Demos addtwoints_server example

Add the service_callback to file:
```
#include "example_interfaces/srv/add_two_ints.h"

void service_callback(const void * req, void * res){
    example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
    example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

    res_in->sum = req_in->a + req_in->b;
}
```
Create and add service to the executor:
```
rcl_service_t service;
rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints");

example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;
rclc_executor_add_service(&executor, &service, &req, &res, service_callback);
```

2. Flash to pico, spawn micro ros agent
3. Call service:
```
ros2 service call /addtwoints example_interfaces/srv/AddTwoInts "a: 10
b: 15"
```

#### Expected behavior
Expect service response sum = 25

#### Actual behavior
No response is received for the service request.

If the pico is attached via a debugger and a breakpoint is set for `service_callback` it will be called, but the contents of req_in is garbage, with the `a` field containing random data and the `b` field equaling 1 rather than the 10 and 15 expected.

Example response from gdb when sending a = 10 and b = 15:
```
(gdb) p *req_in
$5 = {a = -5146191458559808030, b = 1}
```

#### Additional information
I suspect this is what is also breaking the parameter server. If a pameter_server is created and a service is called for that parameter server, the executor will return a failure with the failure occurring somewhere during deserialization.

pablogs9, 2022-04-05:
Can you share your whole code?

rjp5th, 2022-04-05:
Ah, I was not running Fast DDS, when I switched over the services started working. Thanks!

Here's my full code for completeness:
```
#include <stdio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include "pico/stdlib.h"
#include "pico_uart_transports.h"

const uint LED_PIN = 25;

#include "example_interfaces/srv/add_two_ints.h"

void service_callback(const void * req, void * res){
    example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
    example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

    res_in->sum = req_in->a + req_in->b;
}

int main()
{
    rmw_uros_set_custom_transport(
		true,
		NULL,
		pico_serial_transport_open,
		pico_serial_transport_close,
		pico_serial_transport_write,
		pico_serial_transport_read
	);

    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);

    rcl_timer_t timer;
    rcl_node_t node;
    rcl_allocator_t allocator;
    rclc_support_t support;
    rclc_executor_t executor;

    allocator = rcl_get_default_allocator();

    // Wait for agent successful ping for 2 minutes.
    const int timeout_ms = 1000; 
    const uint8_t attempts = 120;

    rcl_ret_t ret = rmw_uros_ping_agent(timeout_ms, attempts);

    if (ret != RCL_RET_OK)
    {
        // Unreachable agent, exiting program.
        return ret;
    }

    rclc_support_init(&support, 0, NULL, &allocator);

    rclc_node_init_default(&node, "pico_node", "", &support);

    rcl_service_t service;
    rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints");

    rclc_executor_init(&executor, &support.context, 1, &allocator);

    example_interfaces__srv__AddTwoInts_Response res;
    example_interfaces__srv__AddTwoInts_Request req;
    rclc_executor_add_service(&executor, &service, &req, &res, service_callback);

    gpio_put(LED_PIN, 1);
    while (true)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
    }
    return 0;
}
```

closed 2022-04-05 by rjp5th