micro-ROS/micro_ros_raspberrypi_pico_sdk #716: where is microros in CMakeLists.txt

tianrking, 2022-11-22:
I would like to know which is the corresponding file for this.  I could not find it.

libraries microros  in below code 

```c++
target_link_libraries(pico_micro_ros_example
    pico_stdlib
    microros
)
```

pico_stdlib is in pico_sdk/src/common/ but where is microros

tianrking, 2022-11-23:
@pablogs9 thanks for your reply. but I dont understand why cannot write like this

```c++
target_link_libraries(pico_micro_ros_example
    pico_stdlib
    libmicroros
)
```

why microros instead of libmicroros .
and in libmicroros DIR i cant find something like microros.xxx

pablogs9, 2022-11-23:
https://stackoverflow.com/questions/6561273/is-liblibrary-name-a-so-a-naming-convention-for-static-libraries-in-linux

closed 2022-11-23 by pablogs9