micro-ROS/micro_ros_raspberrypi_pico_sdk #765: pico_micro_ros_example does not connect to microros/micro-ros-agent:humble as described in readme

KySmith1, 2023-01-30:
####  Summary
It appears the current version of pico_micro_ros_example isn't functional with the latest micro-ros-agent:humble container. I've attempted all the troubleshooting from https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/26, but come up empty handed. 

####  Environment & Setup:
* Rasperry Pi Pico connected via known good USB cable
* Docker 20.10.23 on Ubuntu 22.04.1 LTS 
* Using current `microros/micro-ros-agent:humble` container
(sha256:2e224157620639814f988ae3f20d1edefa2b06f9bc0e2574421edb1418030e01)
* Using `humble` branch at the time of writing 
(commit: e87995c3f3f4e4bbd1847bae9d25adca2b1de795)

#### Steps to reproduce the issue
1. Compile and deploy microros agent as described in the readme. Output shows build was successful. Pico mass storage device unmounted from the Ubuntu OS as expected.
2. Start micro-ros-agent via the following command:
```sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyACM0 -b 115200```
3. Two messages appear in the container's stdout, but a session is never established. Below is an example of the current behavior.
[image]
`minicom` observes something happening at `/dev/ttyACM0` which I'm assuming is the polling from `rmw_uros_ping_agent`. Every more or less two seconds, something happens in a minicom window:
```minicom --device /dev/ttyACM0```

#### Expected behavior
ROS docker container communicates a session has been established and messages from the `pico_publisher` message are recorded as seen here: https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/26#issuecomment-825601450

pablogs9, 2023-01-30:
Could you share what do you see with minicom in `/dev/ttyACM0 `?
Also, could you check if using `/dev/serial/by-id/...` path it behaves different?

KySmith1, 2023-01-30:
#### Minicom Output
Sure- here's what I have on `/dev/ttyACM0`:
[image]

Note that `mincom` is sometimes recording tabs, newlines, spaces, arrow key movements, etc.

#### Path
The behavior appears unchanged when using the alternate path.
[image]

#### Toolchain Version
I'm currently using the latest `gcc-arm-none-eabi` (15:10.3-2021.07-4). Can you confirm that newer versions of the toolchain are compatible?
```
kyle@kyle-uubuntu:~/workspace/micro_ros_raspberrypi_pico_sdk/build$ apt-cache policy gcc-arm-none-eabi
gcc-arm-none-eabi:
  Installed: 15:10.3-2021.07-4
  Candidate: 15:10.3-2021.07-4
  Version table:
 *** 15:10.3-2021.07-4 500
        500 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
        100 /var/lib/dpkg/status
```

pablogs9, 2023-01-30:
Are you opening minicom at 115200 bauds?

Also, note this in the README.md: `micro-ROS precompiled library is compiled using arm-none-eabi-gcc 9.2.1`
Can you check with this version?

KySmith1, 2023-01-30:
I didn't in my first post, but the screenshot [above](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/765#issuecomment-1408151026) was posted with: ```sudo minicom --device /dev/ttyACM0 -b 115200```

I previously was building with 10.XX. If this package is neither forward nor backwards compatible, the readme should have direction to install from a set of tars given this is for a known target rather than ```sudo apt install ... gcc-arm-none-eabi ...```. It doesn't look like 9.XX is available much of anywhere in terms of debians on apt.

I've grabbed 9.2.1 ([reference](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads/9-2-2019-12)) and built. Behavior is a bit different in minicom, but no revelations in terms of the docker container. I'm still seeing the same behavior.

I've posted the output below. Note the reference to 9.2.1.
```
kyle@kyle-uubuntu:~/workspace/micro_ros_raspberrypi_pico_sdk/build$ echo "export PICO_TOOLCHAIN_PATH=~/workspace/toolchain/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin" >> ~/.bashrc
kyle@kyle-uubuntu:~/workspace/micro_ros_raspberrypi_pico_sdk/build$ source ~/.bashrc
kyle@kyle-uubuntu:~/workspace/micro_ros_raspberrypi_pico_sdk/build$ PICO_SDK_PATH=~/workspace/pico/pico-sdk cmake .. -DPICO_TOOLCHAIN_PATH=/home/kyle/workspace/toolchain/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin
Using PICO_SDK_PATH from environment ('/home/kyle/workspace/pico/pico-sdk')
PICO_SDK_PATH is /home/kyle/workspace/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/kyle/workspace/toolchain/gcc-arm-9.2-2019.12-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/kyle/workspace/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
TinyUSB available at /home/kyle/workspace/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
cyw43-driver available at /home/kyle/workspace/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/kyle/workspace/pico/pico-sdk/lib/lwip
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kyle/workspace/micro_ros_raspberrypi_pico_sdk/build
```
The make output can be found below.
```
kyle@kyle-uubuntu:~/workspace/micro_ros_raspberrypi_pico_sdk/build$ make
Scanning dependencies of target bs2_default
[  1%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
[  2%] Built target bs2_default
[  3%] Generating bs2_default.bin
[  4%] Generating bs2_default_padded_checksummed.S
[  4%] Built target bs2_default_padded_checksummed_asm
[  5%] Creating directories for 'ELF2UF2Build'
[  6%] No download step for 'ELF2UF2Build'
[  7%] No update step for 'ELF2UF2Build'
[  8%] No patch step for 'ELF2UF2Build'
[  9%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kyle/workspace/micro_ros_raspberrypi_pico_sdk/build/elf2uf2
[ 10%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[ 12%] No install step for 'ELF2UF2Build'
[ 13%] Completed 'ELF2UF2Build'
[ 13%] Built target ELF2UF2Build
Scanning dependencies of target pico_micro_ros_example
[... 79 lines cut ...]
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kyle/workspace/micro_ros_raspberrypi_pico_sdk/build/pioasm
[ 97%] Performing build step for 'PioasmBuild'
[ 10%] Building CXX object CMakeFiles/pioasm.dir/main.cpp.o
[ 20%] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.o
[ 30%] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.o
[ 40%] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.o
[ 50%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[ 60%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[ 90%] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.o
[100%] Linking CXX executable pioasm
[100%] Built target pioasm
[ 98%] No install step for 'PioasmBuild'
[100%] Completed 'PioasmBuild'
[100%] Built target PioasmBuild
```
Is this working for you locally on `humble`?

pablogs9, 2023-01-30:
I will check

KySmith1, 2023-01-30:
Thanks- for what it's worth, I'm seeing the same issue with the `foxy` branch, so I bet it's still something on my end.

Let me know if I can help out at all.

Acuadros95, 2023-03-06:
Closing as the proposed solution should fix the issue.

Feel free to reopen if the problem persist.

closed 2023-03-06 by Acuadros95