micro-ROS/micro_ros_raspberrypi_pico_sdk #770: Pi Pico W support ?

DaFray31, 2023-02-02:
I tried to install on Pi pico W, but when running the example, the led does not work. 

I tried changing the value of the LED without success:
```
PICO_DEFAULT_LED_PIN
WL_GPIO0
```

Agent :
```
[1675349054.035242] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1675349054.038604] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

Is microros compatible with Pi pico w ?

pablogs9, 2023-02-02:
Your micro-ROS client is not communicating with the Agent.

It shall be compatible but could you detail your procedure in order to see what is wrong?

DaFray31, 2023-02-02:
I started with the wrong agent tty, it now works for the agent, but the LED does not work.

```

[1675351131.080451] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1675351131.111944] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1675351131.632653] info     | Root.cpp           | create_client            | create                 | client_key: 0x0C007307, session_id: 0x81
[1675351131.637660] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0C007307, address: 0
[1675351132.214709] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x0C007307, participant_id: 0x000(1)
[1675351132.220233] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0C007307, topic_id: 0x000(2), participant_id: 0x000(1)
[1675351132.223710] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0C007307, publisher_id: 0x000(3), participant_id: 0x000(1)
[1675351132.227425] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0C007307, datawriter_id: 0x000(5), publisher_id: 0x000(3)
```

pablogs9, 2023-02-02:
Sorry but handling the LED is out of the scope of micro-ROS

closed 2023-02-02 by pablogs9

DaFray31, 2023-02-02:
Well, I notice that even the RPI Pico W documentation only mentions the LED at the micropython level...

Are you thinking of adding micro-ros and micropython compatibility by any chance ?

Thanks for the quick answers