micro-ROS/micro_ros_raspberrypi_pico_sdk #872: General question regarding executor behavior

liamhan0905, 2023-04-21:
## Issue template

- Hardware description: Raspberry Pi 4 + Pico

It seems like mico_ros_raspberrypi_pico_sdk doesn't support multithreading.So when a new msg is the received, does the executor call the subscriber and block the main, since they're both on the same thread? I've been getting weird behaviors of the robot when using the callback. Without the callback, I can call methods like `robot.moveFoward()`. But with the subscriber, the same method will only move some of the motors.

```
    // create publisher
    rclc_publisher_init_default(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
        "pub");

    // create subscriber
    rclc_subscription_init_default(
        &subscriber,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
        "robot/cmd_vel");

    // create timer
    rclc_timer_init_default(
        &timer,
        &support,
        RCL_MS_TO_NS(1000),
        timer_callback);

    send_msg.data = 0;
    recv_msg.data = 0;

    rclc_executor_init(&executor, &support.context, 2, &allocator);
    rclc_executor_add_timer(&executor, &timer);
    rclc_executor_add_subscription(&executor, &subscriber, &recv_msg, &subscription_callback, ON_NEW_DATA);
    
     while (true)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
        
        // Do Something. E.g. Moving the robot depending on the Twist msg type
        
            
    }

```

pablogs9, 2023-04-21:
Hello @liamhan0905 micro-ROS does not create threads, you should no block the execution on the subscription callback. Instead notify you application.

CC: @JanStaschulat

JanStaschulat, 2023-04-21:
Hello @liamhan0905, micro-ROS does not create any threads. (neither in RCLC with the Executor nor in the XRCE-DDS client middleware). You mentioned in the description, that only one motor is updated, when using Executor/Subscription. In your code snippet I see only one motor. Could you extend the example or give a link to the repo with the source code? I  did not understand the issue: is is an additional delay - or that "the other motor" which is not part of this example,does not work at all?

For understaning: 

```C
 rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100))
```

blocks upto 100ms if no new messages arrive. But when a new message was received, your `subscription_callback` is immediately called. There are no blocking times between the timepoint when a new message is received and the callback is processed, and the code block: 

```C
        if (robotState == RobotState::MOVE_FORWARD)
        {
                robot.setSpeed(0.4);
                robot.moveForward();
        }
        else if (robotState == RobotState::MOVE_BACKWARD)
        {
                robot.setSpeed(0.4);
                robot.moveBackward();
        }
```

The timing should be equivalent, when you move the if-else-statment to the `subscription_callback`.

JanStaschulat, 2023-04-21:
Also be aware, that the spin_some with timeout 100ms is not calling the Executor/or the other code with a fixed time period of 100ms. The spin_some function may return earlier, if a new message was received. I don't know if that was also your intend. 

If you want periodic motor updates, that it would be better to create a timer with timeout 100ms. Then in the subscription callback you set the state. In the timer callback you set the motor speed and direction. This way you have de-coupled asynchronous message arrival and fixed periodic control loop. 

In this case, rclc is single-threaded. So you don't need any mutex/lock. But if you would run this example in multiple threads you would need some mutex/locks, because you would share data between the subscription callback and the timer callback.

liamhan0905, 2023-04-21:
I like your suggestion of putting the motor control inside the timer callback. Also, the `robot.setSpeed` method sets the PWM speed for each 4 motors. Here are the source code if it helps ([motorControl.cpp](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/50fbdf414fc2608826804ccb1a9ce310bb39ca5e/src/motorControl.cpp#L47-L50), [robotControl.cpp](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/50fbdf414fc2608826804ccb1a9ce310bb39ca5e/src/robotControl.cpp)). I confirmed that the methods (`robot.moveForward()` or `robot.moveBackward()`) work well because

Inside the while loop, if I call the `robot.moveForward()` or `robot.moveBackward()` method, all 4 motors move.
```
    while (true)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(500));
        robot.setSpeed(0.5);
        robot.moveForward();
        sleep_ms(2000);
        robot.moveBackward();
        sleep_ms(2000);
    }
```
But if I comment out the moveForward, moveBackward part from the while loop and move those into the timer_callback as shown below, I still get the similar issue of just one of the motor (motor4) spinning. Here's the [full source code](https://github.com/liamhan0905/micro_ros_raspberrypi_pico_sdk/blob/liam/twist_subscriber/main.cpp#L60-L73). I've been banging my head trying to debug the issue but hitting a wall.

```
void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  // just setting a fixed speed for now.
  if (robotState == RobotState::MOVE_FORWARD)
  {
        robot.setSpeed(0.4);
        robot.moveForward();
  }
  else if (robotState == RobotState::MOVE_BACKWARD)
  {
        robot.setSpeed(0.4);
        robot.moveBackward();
  }
}

....

    while (true)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(500));
//      robot.setSpeed(0.5);
//      robot.moveForward();
//      sleep_ms(2000);
//      robot.moveBackward();
//      sleep_ms(2000);
    }
    return 0;

```

JanStaschulat, 2023-08-23:
Great, that it is now working.

closed 2023-08-23 by JanStaschulat