micro-ROS/micro_ros_raspberrypi_pico_sdk #958: pico_micro_ros_example.uf2 was not created

Russ76, 2023-08-29:
## Issue template

- Hardware description: 
- Celeron mini desktop with Jammy and Humble
- RTOS: 
- Trying to program Raspberry Pi Pico
- Installation type: 
- MicroROS is installed and runs
- Version or commit hash: 
- Humble

#### Steps to reproduce the issue

micro_ros_raspberrypi_pico_sdk this folder wasn't created by install script
Only one was pico-sdk in the Home folder, with 10 folders under it

#### Expected behavior
Could not find expected example file to upload after compiling

#### Actual behavior
Much compiling was done but nothing by this name was created: pico_micro_ros_example.uf2

#### Additional information
Which compiler folder is desired for the path statement? The program seemed to install several places.
Could this be the cause of error?

pablogs9, 2023-09-04:
Could you share the build log and explain the procedure you are following?

Jannled, 2023-10-31:
I might know what could have gone wrong, since I have stumbled upon the same pitfall. It is never mentioned in the getting started guide, that you actually have to clone this git repo. While this might seem obvious, people who are just following along could miss this.