micro-ROS/micro_ros_raspberrypi_pico_sdk #1024: Not enough memory in the buffer stream while running example code

robotaimo, 2023-11-23:
Hey guys,

I'm working on ROS2 Humble with the pico pi and its corresponding micro ros version whereby I also have the following behavior with ROS2 Iron. (Host machine is Intel I5, with Ubuntu 22.04)

I followed the installation instructions of the repo and built the firmware. 
Everything works like a charm.

As soon as I flash the uf2 file, which is generated when you compile the example code, to the pico pi, the LED starts to light up as expected, but I get the following error:

`terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
Aborted`

Additional behavior: 

The very first time I connect the rp2040  to the host machine, and enter `ros2 topic list` I can see /pico_publisher. 
When I try to echo to this topic I get a different error message in the terminal that runs the micro-ros-agent: 
`terminate called after throwing an instance of 'std::bad_alloc'
what(): std::bad_alloc
Aborted`

Please note: this error message only appears once after the initial start of the picos... otherwise i only see the one above
I tried different controllers, different usb-ports, other ROS versions (as said before...). I don't have any ideas anymore...

That's why I'm really infinitely grateful for any advice you can give me!!!

pablogs9, 2023-11-23:
Does it work if you use the same version (Iron) in the micro-ROS client, agent and ROS 2?

robotaimo, 2023-11-23:
Not sure if I got you right:

I am using ROS2 humble on the host Machine and cloned the humble branch of this micro_ros_pico_sdk repo.

pablogs9, 2023-11-23:
You are using the provided example? or other code? if so, could you provide it?

robotaimo, 2023-11-23:
No, just using the provided example on a brand new pico pi.

pablogs9, 2023-11-23:
Could you paste the output of the complete micro-ROS Agent with `-v6` flag?

robotaimo, 2023-11-23:
It works now...  a reboot of my host machine after purging all remaining iron stuff manually and resetting env variables seems to fix it.
My guess is, that I had some artifacts of not corresponding ros versions on my host machine... 
That means that the pico_sdk humble versions works for sure, but maybe there is a bug in the iron branch... because at first I was using Iron and switched than to humble. 
This can be closed now, but maybe have a look on the iron version of the pico_sdk.
Thanks for your help anyway!!

closed 2023-11-23 by pablogs9