micro-ROS/micro_ros_raspberrypi_pico_sdk #1045: Breakpoint Reached On Agent Disconnect Or Failed Ping

808brick, 2023-12-30:
## Issue template

- Hardware description: Raspberry Pi Pico
- RTOS: None
- Installation type: Cloned micro_ros_raspberrypi_pico_sdk Repo
- Version or commit hash: Attempted on iron, humble, and foxy branches

#### Issue Description
Calling the `rmw_uros_ping_agent` function several times, which should return a `rcl_ret_t` data type, and failing to connect to MicroROS Agent after a certain number of tries causes program to exit/crash/breakpoint. 

#### Steps to reproduce the issue
Calling `rmw_uros_ping_agent` function several times (such as when confirming your MicroROS agent connection like [in this micro ros example](https://github.com/micro-ROS/micro_ros_arduino/blob/36194f5cded20e615ce5bfb8b2e890f740be43c9/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) )

#### Expected behavior
A `rcl_ret_t` datatype should be returned that is not of value `RMW_RET_OK`, and the program should be allowed to continue.

#### Actual behavior
Program terminates after a few attempts of running `rmw_uros_ping_agent`. 

#### Additional information
I've attempted this with my own code, as well as with a modified version of the pico_micro_ros_example.c from the micro_ros_raspberrypi_pico_sdk, in which I replace this code:

```
if (ret != RCL_RET_OK)
{
    // Unreachable agent, exiting program.
    return ret;
}
```
with this
```
while (ret != RCL_RET_OK)
{
    ret = rmw_uros_ping_agent(timeout_ms, attempts);
}
```
And still hit the breakpoint with a similar Callstack as shown in the image below:
<hr>

#### Debugger Callstack (using Pi Debugger Probe):

[image]

[Text version of callstack for search &readability]
_exit (pico-sdk runtime.c)
panic (pico-sdk runtime.c)
hw_claim_unused_from_range (pico-sdk claim.c)
user_irq_claim_unused (pico-sdk irq.c)
stdio_usb_init (pico-sdk usb.c)
stdio_init_all (pico-sdk pico_stdio/stdio.c)
pico_serial_transport_open (micro_ros_raspberrypi_pico_sdk pico_uart_transport.c)
uxr_init_custom_transport (unknown source)
rmw_uxrce_transport_init (unknown source)
rmw_uros_ping_agent (unknown source)
initialize_microros (my function)
main (my main function call)

808brick, 2023-12-30:
After several hours of fiddling with it, it seems the issue lies in the fact that any time `rmw_uros_ping_agent` is called, it calls the pico-sdk function `stdio_init_all()`, and calling this several times eg:

```
while (true){
    stdio_init_all();
}
```

Leads to a similar callstack crash. In the `micro_ros_raspberrypi_pico_sdk` repo folder, there is the code for the pico uart transports in `pico_uart_transport`, and has the following function definition:

```
bool pico_serial_transport_open(struct uxrCustomTransport * transport)
{
    stdio_init_all();
    return true;
}
```
Which is one of the functions which show up in the debugger callstack before the breakpoint happens (it gets run any time `rmw_uros_ping_agent` is run). I'm assuming this was shoved in as a means to keep most of the MicroROS bare example code universal, instead of doing what "should" be done, which is calling `stdio_init_all();` near the beginning of the main() function call (only once).

If I comment out the `stdio_init_all();` function in the source file (since I cloned the repo) like so:

```
bool pico_serial_transport_open(struct uxrCustomTransport * transport)
{
    // stdio_init_all();
    return true;
}
```

And rebuild the MicroROS pico library

```
docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:humble
```

AND ensure to run the `stdio_init_all` function at some point before I call my `rmw_uros_ping_agent`, it seems I can run the function as many times as I want without crashing/hitting the breakpoint.

I'm going to close this issue for now, but this may be worth addressing / doing a PR for which removes the `stdio_init_all` call from `pico_serial_transport_open` and instead is called in the example main functions directly, as this does not seem like the "correct" way of doing it (either that or there is something other than `rmw_uros_ping_agent` which I should be using for the purpose of checking my agent connection, but as mentioned in my original post, that is the way it is shown in examples). Or there should be some type of initalization flag or ifdef which ensures that `stdio_init_all` does not get repeatedly called.

closed 2023-12-30 by 808brick

pablogs9, 2024-01-05:
@808brick thoughts on this: https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/pull/1048 ?

808brick, 2024-01-06:
By the looks of it this should address the issue for most users in a way that keeps most of the MicroROS examples valid. 

Some experienced Pico users might still run the `stdio_init_all` function in their main code before there `pico_uart_transport.c` code does it again, not realizing it's being run for them, but the Pico should be able to handle 2 inits just fine. (eg: most developers initialize IRQ handlers or GPIO before setting up ROS entities then spinning)

When I get a chance I will clone a fresh repo and ensure the problem is resolved and ensure another init call doesn't affect any pre-configured IRQ/IO.