micro-ROS/micro_ros_raspberrypi_pico_sdk #1169: microROS PICO - Publisher node in pico_micro_ros_example.c is not coming up and publishing messages [HUMBLE]

kasunvinod, 2024-05-09:
- Hardware description: Raspberry Pi Pico W 2022
- Installation type: Followed the steps mentioned in the README.md
- Version or commit hash: humble

#### Steps to reproduce the issue
1. Clone Pico SDK, compile it and configure the environment variables.
2. Clone the micro_ros_raspberrypi_pico_sdk for humble branch and compile it.
3. Flash the Pico with pico_micro_ros_example.uf2 file using picotool.
4. Started the microROS agent using the Docker container.
At this point the connection between the pico and the main computer established, the topic and the publisher is created as shown in the below image.
[image]
5. But when I execute the `ros2 topic list` and `ros2 node list` commands, nothing is coming up.
[image]

#### Expected behavior
1. pico_node should be active in the node list,
2. pico_publisher should be active in the topic list.
3. A number should be published to the above topic.

#### Actual behavior
1. No pico_node
2. No pico_publisher

#### Additional information
1. Main computer is running Ubuntu 22.04 and ROS2 Humble.
2. RMW Implementation is `rmw_cyclonedds_cpp`.
3. Initially I build the example code with arm-none-eabi-gcc 10.3.1 version (it had the same issue) then I re-build it using 9.3.1. Attached screenshots are related to version 9.3.1 build.
4. Unplugged and re-plugged the pico multiple times but the issue is still there.

kasunvinod, 2024-05-09:
@pablogs9 Thank you for the fast reply. I will check this and get back to you. 

I'm having another issue related to microROS STM32 library. It is posted in the STM32 microROS repo [Issue 143](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/143). I really appreciate it if you could give me feedback about it as well.

kasunvinod, 2024-05-10:
Changed the DDS back to fastrtps but the issue is still there.

[image]

kasunvinod, 2024-05-10:
This is what I can see when i listen to the port using minicom. Some characters are continuously printed on the terminal.

[image]

pablogs9, 2024-05-10:
Can you paste here the output of the micro-ROS Agent using the flag `-v6`?

kasunvinod, 2024-05-10:
> admin_ka@robotcs:~$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM1 -v6
[1715321602.723526] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1715321602.723651] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1715321607.985504] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1715321607.995604] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM1, error 2, waiting for connection...
[1715321609.001682] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM1, error 2, waiting for connection...
[1715321609.982873] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1715321610.556986] info     | Root.cpp           | create_client            | create                 | client_key: 0x0C0AF7DB, session_id: 0x81
[1715321610.557019] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0C0AF7DB, address: 0
[1715321610.557153] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1715321610.559261] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 40, data: 
0000: 81 80 00 00 01 07 20 00 00 0A 00 01 01 03 00 00 12 00 00 00 00 01 00 10 0A 00 00 00 70 69 63 6F
0020: 5F 6E 6F 64 65 00 00 00
[1715321610.563870] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x0C0AF7DB, participant_id: 0x000(1)
[1715321610.563989] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1715321610.563996] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1715321610.565061] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1715321610.568081] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 80, data: 
0000: 81 80 01 00 01 07 46 00 00 0B 00 02 02 03 00 00 38 00 00 00 12 00 00 00 72 74 2F 70 69 63 6F 5F
0020: 70 75 62 6C 69 73 68 65 72 00 00 01 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64
0040: 64 73 5F 3A 3A 49 6E 74 33 32 5F 00 00 01 00 00
[1715321610.568125] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0C0AF7DB, topic_id: 0x000(2), participant_id: 0x000(1)
[1715321610.568147] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1715321610.568154] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1715321610.569060] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1715321610.569852] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1715321610.569874] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0C0AF7DB, publisher_id: 0x000(3), participant_id: 0x000(1)
[1715321610.569982] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1715321610.569989] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1715321610.571064] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1715321610.572066] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 36, data: 
0000: 81 80 03 00 01 07 1C 00 00 0D 00 05 05 03 00 00 0E 00 00 00 00 02 01 10 03 00 01 00 0A 00 00 00
0020: 00 00 00 03
[1715321610.572169] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0C0AF7DB, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1715321610.572281] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1715321610.572288] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1715321610.573103] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 13, data: 
[... cut ...]
0000: 81 80 10 00 07 01 08 00 00 1A 00 05 0C 00 00 00
[1715321623.574073] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data: 
0000: 0C 00 00 00
[1715321623.574186] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0C0AF7DB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 11 00 00 00 80
[1715321624.574011] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0C0AF7DB, len: 16, data:

pablogs9, 2024-05-10:
Your micro-ROS Client is correctly publishing to the DDS dataspace (check those `<<DDS>>` that means an output DDS message).

In this case, maybe the problem is on the DDS side, are you running on the correct ROS 2 Domain ID? Can you check something like `ROS_DOMAIN_ID=0 ros2 topic list` ??

kasunvinod, 2024-05-10:
@pablogs9 Yes it was a ROS_DOMAIN_ID related issue. I've included a different domain id in the bashrc as I was doing some experiments using TurtleBot. After changing it back to 0, everything works fine now. 

Could you please explain me why it should be ROS_DOMAIN_ID=0? And is there a way to get the same result in a different domain id?

pablogs9, 2024-05-10:
You can change the used domain id in the micro-ROS code, by default it uses the Domain ID 0, but it can be chosen: 

https://github.com/micro-ROS/micro-ROS-demos/blob/d4066c5fd47d520451eefda1fdbeb01e7c206aec/rclc/configuration_example/configured_publisher/main.c#L49

kasunvinod, 2024-05-10:
Thank you so much! I've been trying to find the issue for weeks. You are a legend!

closed 2024-05-10 by kasunvinod