micro-ROS/micro_ros_raspberrypi_pico_sdk #1274: Failed   <<< microcdr when using micro_ros_static_library_builder

danielallstar, 2024-10-02:
## Issue template

- Hardware description: 

#### Steps to reproduce the issue
I clone the micro_ros_raspberrypi_pico_sdk library and try to run the following command in the cloned library:
`
docker run -it --rm -v $(pwd):/project microros/micro_ros_static_library_builder:jazzy
`
#### Expected behavior
I expect that the static library gets recompiled and libmicroros gets rebuild
#### Actual behavior
The compilation stops at microcdr. IN the error message it seems that some files cannot be found.

#### Additional information
The error I get in the terminal

--- stderr: microcdr                                                                                                                                                                     
CMake Error at /project/microros_static_library/library_generation/toolchain.cmake:1 (include):
  include could not find requested file:

    /pico-sdk/cmake/preload/toolchains/find_compiler.cmake
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:53 (project)

PICO_GCC_TRIPLE defaulted to arm-none-eabi
CMake Error at /project/microros_static_library/library_generation/toolchain.cmake:18 (pico_find_compiler):
  Unknown CMake command "pico_find_compiler".
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:53 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
---
Failed   <<< microcdr [0.60s, exited with code 1]
Aborted  <<< rosidl_cli [0.62s]
Aborted  <<< tracetools_trace [0.59s]
Aborted  <<< tracetools_read [0.58s]
Aborted  <<< tracetools [0.59s]
Aborted  <<< rosidl_typesupport_interface [0.62s]
Aborted  <<< test_interface_files [0.61s]
Aborted  <<< rcutils [0.64s]
Aborted  <<< rmw_implementation_cmake [0.60s]

Summary: 0 packages finished [1.02s]
  1 package failed: microcdr
  8 packages aborted: rcutils rmw_implementation_cmake rosidl_cli rosidl_typesupport_interface test_interface_files tracetools tracetools_read tracetools_trace
  6 packages had stderr output: microcdr rcutils rmw_implementation_cmake rosidl_typesupport_interface test_interface_files tracetools
  69 packages not processed
[ros2run]: Process exited with failure 1
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
0 upgraded, 0 newly installed, 0 to remove and 250 not upgraded.
rsync: [sender] change_dir "/project/libmicroros/include/action_msgs/action_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/actionlib_msgs/actionlib_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/builtin_interfaces/builtin_interfaces" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/common_interfaces/common_interfaces" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/composition_interfaces/composition_interfaces" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/control_msgs/control_msgs" failed: No such file or directory (2)
[... 130 lines cut ...]
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/test_ros2trace/test_ros2trace" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/test_tracetools_launch/test_tracetools_launch" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tf2_msgs/tf2_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tracetools/tracetools" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tracetools_launch/tracetools_launch" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tracetools_read/tracetools_read" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tracetools_test/tracetools_test" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/tracetools_trace/tracetools_trace" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/trajectory_msgs/trajectory_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/type_description_interfaces/type_description_interfaces" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/unique_identifier_msgs/unique_identifier_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/libmicroros/include/visualization_msgs/visualization_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]

Dhoward094, 2024-10-23:
I have the same issue when running the iron version of the build container. So this isn't specific to ros2 version.

maax-cha, 2024-11-01:
This is because pico-sdk 2.0 has removed the find_compiler.cmake file.

Alternatively, I simply edited the `microros_static_library/library_generation/library_generation.sh` file.

```
git clone https://github.com/raspberrypi/pico-sdk /pico-sdk
```
to
```
git clone --branch 1.5.1 https://github.com/raspberrypi/pico-sdk /pico-sdk
```

Downgrading the SDK version may not be the correct answer; however, it's working. :+1: